MitjaNemec/PlaceFootprints #1: ModuleNotFoundError("No module named 'PlaceFootprints.initial_dialog_GUI'")

AiGreek, 2022-01-16:
Hi, 

I tried this plugin (without using PCM because I'm not a fan) and when I copy/paste the folder with my other plugins I get an error.
I select on my PCB my first led and when launching the script, it tells me:

[image]

`ModuleNotFoundError("No module named 'PlaceFootprints.initial_dialog_GUI'")`

Then, I deleted the log file, and despite the warning in Kicad, this file is not only not generated, but it does not seem to update during an error (At this moment, log file is empty)

Basically, with my version of kicad, the plugin does not work. Should I install something, in addition to this plugin?

```
Application: KiCad PCB Editor (64-bit)

Version: (6.99.0-462-gb605d6d9ae), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jan 15 2022 21:08:15
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

AiGreek, 2022-01-16:
With PCM install it seems to work (almost) i have this error this time 

```
01-16 07:56:17 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 378:Plugin executed on: 'win32'
01-16 07:56:17 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 379:Plugin executed with python version: '3.9.6 (default, Sep 27 2021, 21:15:11) [MSC v.1929 64 bit (AMD64)]'
01-16 07:56:17 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 380:KiCad build version: (6.99.0-462-gb605d6d9ae)
01-16 07:56:17 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 381:Plugin version: 1.0.0
01-16 07:56:17 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 382:Frame repr: <wx._core.Frame object at 0x0000024225E3C670>
01-16 07:56:17 com_github_MitjaNemec_PlaceFootprints.place_footprints 123:getting a list of all footprints on board
01-16 07:56:21 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 559:Reference designator is: D
01-16 07:56:21 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 560:Reference number is: 1
01-16 07:56:21 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 592:Sorted and filtered list:
['D1', 'D2', 'D3', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'D10', 'D11', 'D12', 'D13', 'D14', 'D15', 'D16', 'D17', 'D18', 'D19', 'D20', 'D21', 'D22', 'D23', 'D24', 'D25', 'D26', 'D27', 'D28', 'D29', 'D30', 'D31', 'D32']
01-16 07:56:28 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 625:Footprints to place:
['D1', 'D2', 'D3', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'D10', 'D11', 'D12', 'D13', 'D14', 'D15', 'D16', 'D17', 'D18', 'D19', 'D20', 'D21', 'D22', 'D23', 'D24', 'D25', 'D26', 'D27', 'D28', 'D29', 'D30', 'D31', 'D32']
01-16 07:56:28 com_github_MitjaNemec_PlaceFootprints.place_footprints 253:Starting placing with circular layout
01-16 07:56:28 com_github_MitjaNemec_PlaceFootprints.place_footprints 263:reference footprint position at: VECTOR2I(175000000, 142000000)
01-16 07:56:28 com_github_MitjaNemec_PlaceFootprints.place_footprints 268:rotation center at: (175000000, 185029000.0)
01-16 07:56:28 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 638:Fatal error when executing place footprints
Traceback (most recent call last):
  File "C:\Users\edoua\OneDrive\Documents\KiCad\6.99\3rdparty\plugins\com_github_MitjaNemec_PlaceFootprints\action_place_footprints.py", line 634, in Run
    placer.place_circular(footprints_to_place, ref_fp_ref, radius, delta_angle)
  File "C:\Users\edoua\OneDrive\Documents\KiCad\6.99\3rdparty\plugins\com_github_MitjaNemec_PlaceFootprints\place_footprints.py", line 274, in place_circular
    fp.fp.SetPosition(pcbnew.wxPoint(*new_position))
  File "C:\Program Files\KiCad\6.99\bin\Lib\site-packages\pcbnew.py", line 17381, in SetPosition
    return _pcbnew.FOOTPRINT_SetPosition(self, aPos)
TypeError: in method 'FOOTPRINT_SetPosition', argument 2 of type 'VECTOR2I const &'
```

MitjaNemec, 2022-01-16:
Whoa, that was fast.

The first error seems to be that you did not put all the .py file in the folder. Specifically, `initial_dialog_GUI.py` was most likely missing. Though for reference, I am focusing my support on PCM delivery and current stable release.

The second error seems to be related with API change. I am not certain though. Can you get the following signature from PCB editor scripting console
```python
import pcbnew
from inspect import signature
signature(pcbnew.FOOTPRINT.SetPosition)
```
On 6.0.1 I get `<Signature (self, aPos: 'wxPoint') -> 'void'>`

AiGreek, 2022-01-16:
Well i downloaded and extracted directly in C:\Users\edoua\OneDrive\Documents\KiCad\6.99\scripting\plugins like all others plugins (well ... one). I'm not the smartest but it can't be possible that i missed a file ^^

here's what i have from scripting console
`<Signature (self, aPos: 'VECTOR2I') -> 'void'>`

MitjaNemec, 2022-01-16:
Yeah, GitHub only holds the source files. And GUI code is generated using WxFormBuilder from `.fbp` files. I have a script that I run when releasing the package `make_a_package.sh`. It runs on linux and probably only on my machine. What you should do is download the released plugin (.zip file) from [Releases](https://github.com/MitjaNemec/PlaceFootprints/releases) page.

As for the second case, it seems I was right. Python API changed. And as 6.99 is in heavy flux, I'll not even try to follow these changes.

If you feel "olimpic" you might want to cast the wxPoint into VECTOR2I so `fp.fp.SetPosition(pcbnew.wxPoint(*new_position))` should become `fp.fp.SetPosition(pcbnew.VECTOR2I(pcbnew.wxPoint(*new_position)))`

MitjaNemec, 2022-01-16:
[This](https://gitlab.com/kicad/code/kicad/-/commit/c4c56de70871a641ad39b830b37e30c83492b25b) is the culprit for you issues

MitjaNemec, 2022-01-16:
So most likely, there will be a lot of places where similar change will be required

AiGreek, 2022-01-16:
I couldn't get the plugin to work or install via github, so I ended up installing it via the PCM and move all the files where i like them to be, and editing the place_footprint.py file with `fp.fp.SetPosition(pcbnew.VECTOR2I(pcbnew.wxPoint(*new_position)))`

Now it works.

closed 2022-01-17 by MitjaNemec