MitjaNemec/PlaceFootprints #3: Option to conserve orientation of components

lazyzero, 2022-01-28:
Hi,

this plugin saved me so much time, but I am often making matrices of WS2812, and often as zick-zack pattern, meaning that every second row has 180° rotated parts. Could you add a checkbox to keep orientation? 

Best Christian

MitjaNemec, 2022-01-28:
It is a neat idea. But it'll have to be expanded to make it a bit more universal. I'll add the option to rotate each n-th footprint around it's anchor for a specific angle.

As I'am porting other footprints, this will have to wait

lazyzero, 2022-01-31:
Yeah, specific angle would also help for designing hexagonal light PCBs. 

My current problem would be every 5th LED rotate by 180°, to produce rows by five LEDs.

MitjaNemec, 2022-02-08:
The feature has been added