MitjaNemec/PlaceFootprints #4: Placer.place_linear() missing 1 required positional argument: 'copy_text_items'

tomek-szczesny, 2022-01-29:
Hello!

Thanks for a plugin that I hope to find extreme useful, but I am unable to get this to work.

I have a schematic with 8 identical sheets (SSS0-7) that contain many components - so it's a basic usage scenario, I believe.

My system is Manjaro x64, and I downloaded the plugin from within the Kicad 6 built-in manager.

Here's the log:

I'll be happy to help with additional information.

```
01-29 17:41:02 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 378:Plugin executed on: 'linux'
01-29 17:41:02 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 379:Plugin executed with python version: '3.10.1 (main, Dec 18 2021, 23:53:45) [GCC 11.1.0]'
01-29 17:41:02 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 380:KiCad build version: (6.0.1)
01-29 17:41:02 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 381:Plugin version: 1.1.0
01-29 17:41:02 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 382:Frame repr: <wx._core.Frame object at 0x7f99aeb5dea0>
01-29 17:41:02 com_github_MitjaNemec_PlaceFootprints.place_footprints 121:getting a list of all footprints on board
01-29 17:41:04 com_github_MitjaNemec_PlaceFootprints.place_footprints 166:constructing a list of sheets suitable for replication on level:'SSS0', file:'solid_state_switch.kicad_sch'
01-29 17:41:04 com_github_MitjaNemec_PlaceFootprints.place_footprints 201:suitable sheets are:[['SSS1'], ['SSS2'], ['SSS3'], ['SSS4'], ['SSS5'], ['SSS6'], ['SSS7']]
01-29 17:41:08 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 454:Footprints to place: ['D403', 'D1103', 'D1003', 'D803', 'D903', 'D703', 'D503', 'D603']
01-29 17:41:08 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 500:Fatal error when executing place footprints
Traceback (most recent call last):
  File "/home/mctom/.local/share/kicad/6.0/3rdparty/plugins/com_github_MitjaNemec_PlaceFootprints/action_place_footprints.py", line 495, in Run
    placer.place_linear(sorted_footprints, ref_fp_ref, step_x, step_y)
TypeError: Placer.place_linear() missing 1 required positional argument: 'copy_text_items'

```

tomek-szczesny, 2022-01-29:
The same happens for other arrangements too.

I modified the python file to make `copy_text_items` an optional argument, but the problem persists.

```
    def place_linear(self, footprints_to_place, reference_footprint, step_x, step_y, copy_text_items = True):
(...)
```

```
01-29 18:22:20 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 500:Fatal error when executing place footprints
Traceback (most recent call last):
  File "/home/mctom/.local/share/kicad/6.0/3rdparty/plugins/com_github_MitjaNemec_PlaceFootprints/action_place_footprints.py", line 495, in Run
    placer.place_linear(sorted_footprints, ref_fp_ref, step_x, step_y)
TypeError: Placer.place_linear() missing 1 required positional argument: 'copy_text_items'
```

I also modified the calling line (495) to add hardcoded `True` as an extra argument, but the same error is still there! This is beyond my comprehension.

MitjaNemec, 2022-01-29:
Thanks for reporting. This is a bug introduced when fixing #2. I've only checked the "by reference" path of the workflow.
You should be able to fix this by adding another argument. But you have to completely close KiCad when you make a change in plugin code, and restart it.

As for the official fix, it'll take a bit, as I'm in the progress of publishing Replicate Layout plugin, and once it is published, I'll publish the fix for this.

MitjaNemec, 2022-01-29:
I've just check, and additional argument is a proper solution. Before the package becomes available through PCM, you might want to download release 1.1.1

tomek-szczesny, 2022-01-29:
Yep, it solves the problem. I was just about to apply proposed fixes, you were a tad bit quicker than me.
Thank you very much!

closed 2022-02-08 by MitjaNemec