MitjaNemec/PlaceFootprints #5: Something wrong when component is about to be flipped

tomek-szczesny, 2022-01-30:
Hi,

I'm heavily testing this tool today :)

I noticed there is a bug in the following case:
- The components are pre-placed by Kicad on Top side by default
- In my hand-placed template I flipped some components to the bottom side
- When invoking PlaceFootprints, it flips the component and moves to the desired location, but rotation is wrong by 180 degrees.
This is easy to solve by pressing "R" twice on each component placed by the script.

However if I invoke the script on a component on a bottom side, and their counterparts have been flipped beforehand by me, error does not occur.

I suspect KiCad has its own unreasonable way of storing orientation data, or perhaps the order of flipping and rotating in the script does matter. I'd guess it should flip, and then rotate.

MitjaNemec, 2022-01-30:
Thanks for reporting. I'll have to setup better testing suite. I've just went through similar issue with "Replicate Layout" plugin (yet to be published) when I added an option to replicate on flipped side.

tomek-szczesny, 2022-01-30:
Nah, I'm sure all bugs will be reported to you, why would you look for them yourself. :)

MitjaNemec, 2022-01-30:
You were right. The plugin placed the footprints first, and then flipped them if needed. I've fixed this, and a new release is available under Releases. But as before, it will take a while, before it is published in PCM

tomek-szczesny, 2022-01-30:
No worries, I simply rotated them into correct positions myself, it wasn't that much of a burden.
Anyway, thanks again for this excellent plugin. My board is complete now, on time. :)

closed 2022-02-08 by MitjaNemec