MitjaNemec/PlaceFootprints #6: Some new error after last update.

AiGreek, 2022-02-07:
Hey, it's been a while :D

I installed this morning the latest version of your plugin via github (and the pcm afterwards) then, the last version of Kicad Nightly and I find myself with this error : 

[image]

```
Application: KiCad PCB Editor (64-bit)

Version: (6.99.0-818-g79c7d55a40), release build

Libraries:
	wxWidgets 3.1.5
	FreeType 2.11.0
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Feb  6 2022 21:40:25
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

Any idea ?

MitjaNemec, 2022-02-08:
Yes. 6.99 has significant changes in python API. I am only supporting the plugins to work on 6.0.x

AiGreek, 2022-02-08:
ow okey, i'll wait then or try hard to do a little fork. Thanks for the answer ;)

MitjaNemec, 2022-02-08:
You'll wait a while, as I'll start supporting 6.99 when V7 is going to be close to release or slightly after.
You can try to [fix](https://github.com/MitjaNemec/PlaceFootprints/issues/1#issuecomment-1013846845) this yourself.

AiGreek, 2022-06-18:
I finally updated this plugin to work with the last nightly.

```
Application: KiCad PCB Editor (64-bit)

Version: (6.99.0-2176-g69f25f35a8), release build

Libraries:
	wxWidgets 3.1.7
	FreeType 2.11.1
	HarfBuzz 4.2.0
	FontConfig 2.14.0
	libcurl/7.83.1-DEV Schannel zlib/1.2.12

Platform: Windows 11 (build 22000), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jun 17 2022 07:52:05
	wxWidgets: 3.1.7 (wchar_t,wx containers)
	Boost: 1.79.0
	OCC: 7.6.0
	Curl: 7.83.1-DEV
	ngspice: 37
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

It was indeed just a lot of fp.fp.SetPosition(pcbnew.VECTOR2I(pcbnew.wxPoint(*new_position))) to paste.
Works like a charm now !

closed 2022-06-18 by AiGreek