MitjaNemec/PlaceFootprints #11: Change default offset to 0/0

tomek-szczesny, 2022-12-31:
Hi,
Still using this tool on the same board, and just a minor request is to change the default offsets to 0 on both x and y.
The reason being, I have to run the script multiple times for each component, and in most cases the array will be created along either x or y.

I'm using a fairly old version of this tool so pardon me if that has already been taken care of.

Thanks again for your contribution to FLOSS. :)

MitjaNemec, 2023-01-03:
For the time being you can comment out (append a hash symbol `#` at the beginning of the line) the following lines: 246, 247, 251, 252

MitjaNemec, 2023-01-17:
I've changed the UI so that there are no defaults set. I'll leave this open as I can maybe set the default values to the previously set values used in this project (but I'll have to save previously used values somewhere)

tomek-szczesny, 2023-01-17:
The reason why I suggested 0,0 is that I placed components along x or y axis (which I suppose is the most common scenario). I pasted one coordinate from clipboard and put "0" at the other. I wished I could skip that last step, as it involved surprisingly many extra hand movements. :)

I guess storing some values in OS' default temporary directory would be acceptable, even if they get wiped out after KiCAD shutdown. Or as a file within project directory, if the plugin environment allows for that. Sadly I don't know much about all this.
Thank you for your help!

tomek-szczesny, 2023-08-09:
Related to #18