MitjaNemec/PlaceFootprints #12: Not working with Kicad 7.0

tionebrr, 2023-02-14:
Hello dear dev,
Currently not working with kicad 7.0. 
Any plan to update ?

MitjaNemec, 2023-02-15:
Can you please provide more details. Did you get any error message? If you did, what message you've got? Which version of the plugin you've got installed? What is your OS version.

eewiz, 2023-02-16:
Not working for me either.
Windows 10, KiCad 7.0.
Version 1.2.8 of the Place Footprints plugin in Plugin and Content Manager.
Just installed KiCad 7.0 which starts out with no plugins installed.
Your install button is grayed out.
Can't install Place Footprints plugin.

Same issue with Archive 3D Models v1.0.5, Install button is grayed out.

For both plugins there is a functional install button in the right-hand pane which produces the error message
"This package version is incompatible with your kicad version or platform."

All for now

MitjaNemec, 2023-02-16:
Thanks for the report.

I've tested the plugin with 6.99 a month or two ago and it worked, but now it fails.

Besides metadata issue, preventing the plugin to be installed, there are some changes to be done.

It will take a week or two before I find the time to work on this

tionebrr, 2023-02-17:
Thanks for the detailed report @eewiz, and thanks for your time @MitjaNemec .

MitjaNemec, 2023-02-21:
I've just released plugin version 2.0.0 which should work with KiCad V7

Martinnnnnnnnnnn, 2023-02-25:
Just installed 2.0.0, does not work with KiCad 7.
"Fatal error when restoring layout". 
Application: KiCad PCB Editor x64 on x64

Version: (7.0.0), release build

Libraries:
	wxWidgets 3.2.1
	FreeType 2.12.1
	HarfBuzz 5.0.1
	FontConfig 2.14.1
	libcurl/7.83.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Feb 12 2023 01:35:19
	wxWidgets: 3.2.1 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.2
	Curl: 7.83.1-DEV
	ngspice: 39
	Compiler: Visual C++ 1934 without C++ ABI

Build settings:
	KICAD_SPICE=ON

MitjaNemec, 2023-02-25:
Thanks for reporting this. Is it possible for you to attache the `place_footprints.log` file, which should be in the project folder?

Martinnnnnnnnnnn, 2023-02-25:
Sure... but where is the project folder?

MitjaNemec, 2023-02-25:
the folder where you project is located

MitjaNemec, 2023-02-25:
If you think that I could have named this better don't hold back. I am the first one to acknowledge that UX could/should be improved

Martinnnnnnnnnnn, 2023-02-25:
I see... I wondered where the plugin data was...
[place_footprints.log](https://github.com/MitjaNemec/PlaceFootprints/files/10832085/place_footprints.log)

Martinnnnnnnnnnn, 2023-02-25:
I can also upload the project, nothing secret.

MitjaNemec, 2023-02-25:
Please do

Martinnnnnnnnnnn, 2023-02-25:
[VecSchalt.zip](https://github.com/MitjaNemec/PlaceFootprints/files/10832090/VecSchalt.zip)
Here you go. Thanks btw for the fantastic replicate layout - this design is structured (2 levels hierarchy) so do an automated matrix placement using replicate layout. Unfortunately, the first step (place footprints) does not work.

MitjaNemec, 2023-02-25:
The fix should be in [2.0.1](https://github.com/MitjaNemec/PlaceFootprints/releases/tag/2.0.1). You can install it from file or wait a day or so when it should be available via PCM.

I'd appreciate the confirmation from your side

Martinnnnnnnnnnn, 2023-02-25:
Installed it from file - does not crash.
But It doesn't work for me... Selecting SW1 (top left on PCB), place by sheet nr, Hierarchy level Zeile (highlights eight instances correctly), but then does nothing (linear placement). Am I doing something wrong?

MitjaNemec, 2023-02-26:
Ok, se there were two issues present. And I should have fixed the second one in 2.0.2 which is available under Release.

Thanks for testing.

Martinnnnnnnnnnn, 2023-02-26:
Works better - however the component order seems pretty random:
[image]
Here I selected the "Zeile1" block, which consists of eight "TransSch" subblocks (one switch and transistor each). They should be all in the same column (or row, doesn't matter). However all components seem distributed randomly.
What do I need to do to achieve this correct matrix placement?

MitjaNemec, 2023-02-26:
Thanks for confirmation

I assume it is because even when working "by sheet" plugin sorts footprints by reference. But I'd need to confirm this.

If this is true, then we have a dilemma. In your case it would be much better to sort them by sheet number. As your sheets are numbered. 

But there is no guaranty that sheets have to be numbered. With footprints, this is a hard requirement set by KiCad.

This is probably the reason why I'd rather leave things as they are. But I am willing to discuss this

Martinnnnnnnnnnn, 2023-02-26:
There's another bug - related? https://forum.kicad.info/t/v7-hierarchy-connectivity-bug/40541

So to summarize: There currently is no option to place a 2 level matrix correctly:
- Place by sheet number works correctly from your side, but KiCad messes up the sheet numbering (and consequently the annotation), so the order is mixed (see my forum post)
- Place by reference does not work as it does not traverse both hierarchy levels. I assume this is a PlaceFootprints bug?

MitjaNemec, 2023-02-26:
Regarding sorting, i might be able to add a radio button so that the user can select whether to sort footprints with regard to the footprint number or with regard to the sheet name. But with nested hierarchies there are at least two options. sort by shallow hierarchy first and then by go deeper and deeper or vice versa? And then it is up to the user to name hierarchical pages properly. But this will take some time to implement, so I don't recommend waiting with your current design.

Martinnnnnnnnnnn, 2023-02-26:
Aren't these two options already there?
[image]
As I wrote the first one does not work.
BTW initially the window looks like this:
[image]
Just a cosmetic issue.

MitjaNemec, 2023-02-26:
Hmmm this might be a bit complicated so please bear with me.

The "Place by reference nr." option  just looks at the footprint reference and will place footprints which are consecutively numbered with respect to currently selected one. Any hierarchical info (metadata) is disregarded. And when the footprint selection is ready to place, the plugin sorts this list by reference number.

The "Place by sheet nr." will place only (and only) footprints from the same sheet as is the currently selected one. It looks at the hierarchical metadata to figure out design hierarchy. Once the sheet selection is ready, plugin figures out which Footprints it needs to place and sort these footprint by the reference number. Obviously this mode might need a better name. And here I might be able to add the option to sort the footprints by sheetname instead.

As for your design, I'd first place the inner sheet (Zeile1:Transch1-8) and then I'd place anchors for Zeile(1-7). Then I'd use replicate layout to just replicate placement of Zeile1 to Zeile2-7

MitjaNemec, 2023-02-26:
I really appreciate this discussions. It is way better than [rubber duck debuging](https://en.wikipedia.org/wiki/Rubber_duck_debugging) as it helps me think about how plugins should cover extreme cases

Martinnnnnnnnnnn, 2023-03-11:
I have a new project where I have a matrix of transistors and jumpers. I hope I can place those with PlaceFootprint/ReplicateLayout.
However, I can't get it to work. Sheet order seems to be OK, but the plugin picks the wrong component out of the hierarchy.
[VecMatrix.zip](https://github.com/MitjaNemec/PlaceFootprints/files/10949163/VecMatrix.zip)
[image]
In the image, you should see Q101 followed by Q101+n*32, however the references are wrong. Any idea?

MitjaNemec, 2023-03-13:
Thanks for the project. I'll look into it, but it is really really hard for me to find the time. So please be patient

Georges-SR, 2023-09-21:
> Thanks for the project. I'll look into it, but it is really really hard for me to find the time. So please be patient

I know it's not really the place, but I just came to say, a thousand thank you dear sir !! 
Your plugin saved me right now for a very important project (a custom hardware for a French team participating in the international genetically engineered machines competition - iGEM).
Know your work is extremely appreciated, again thank you.
Wishing you all the best !

livingcreative, 2023-11-21:
It does not work for me also. It only disdplays ONE selected footprint in the list and nothing else. There's no reason to place only one footprint, I need to place several.

livingcreative, 2023-11-21:
[image]