MitjaNemec/PlaceFootprints #13: Feature request: Place components in triangular grid.

GrootOpperhoofd, 2023-02-24:
Great feature overall, so far it works great with kicad 7.0. For a new project I have a feature request:

It would be really convenient if I could place components in a triangular grid.
Hexagonal grids would be nice as well.

A rudimentary but easy way to implement would be to skew a rectangular grid and have the option to skip every other point.
When alternating the skipped grid point for each row that would create a triangular grid.

MitjaNemec, 2023-02-24:
Well you'll have to be more verbose to get your point across. Maybe add some sketch?

I don't see how you would not be able to achieve triangular or hexagonal placement if you'd just choose three/six footprints and place them circular with 120˚/60° angle.

Alloyed, 2024-12-27:
I've been doing this manually, but I wouldn't mind this feature!

Here's what I have in mind (+ [the wikipedia article](https://en.m.wikipedia.org/wiki/Circle_packing#Densest_packing)) :

[image]

This picture is in what I've been calling "staggered rows" order, but you can rotate the whole thing 90 degrees and get "staggered columns" instead. this is visualized using circles, but you could replace each circle with a regular hexagon with the same radius, without changing the grid pattern. (since footprints decide what their shape is, this doesn't really matter. but I'm using it for circles.)

the math for actually placing hexagonal grids like this is [visualized here](https://www.redblobgames.com/grids/hexagons/#spacing), but if you already have a 2d grid of elements you can skip to the [algorithms here](https://www.redblobgames.com/grids/hexagons/#hex-to-pixel-offset)

The main benefit of having grids like this is that you can pack more circles into the same space. I use this to get dense grids of potentiometers on panels, while still leaving enough room around each for the user to comfortably fiddle with them. Placing them manually is fine, but since I already know from the schematic that (for example) I have an 8x8 grid to place, having a preset for that would be handy.

I'm not sure what the original poster had in mind for "triangle grids", but you can take a hex grid and have it form the outline of a triangle by adding one element each row, (so 1, then 2, then 3 then 4 etc.)