MitjaNemec/PlaceFootprints #14: Got error while trying to place footprints.

codeman256, 2023-03-17:
Tried any which way and none ever works. attaching log. 
[place_footprints.log](https://github.com/MitjaNemec/PlaceFootprints/files/11006562/place_footprints.log)

MitjaNemec, 2023-03-19:
Thanks for reporting this.

From the error log it looks like that there is a conflict between my plugin and [pcbnewTransition](https://github.com/yaqwsx/pcbnewTransition). As my plugin works on pure KiCadV6 installation I think that this is actually an issue with pcbnewTransition. Try running the plugin again after you've removed pcbnewTransition.

closed 2023-03-19 by MitjaNemec

codeman256, 2023-03-19:
Oh okay. I'm guessing that pcbnewTransition is a requirement for one of the other plugins I have because I don't see it in my list. do you know which one? 
[image]

MitjaNemec, 2023-03-19:
I can only confirm that "Archive 3D Models" and "Place Footprints" don't use it. I employed similar approaches during V5/V6 transition, but I've learned my lesson. I assume that "Interactive Html Bom" also doesn't use it as author "qu1ck" is the one who directed me off the monkeypatching. 

You should be able if you search for "pcbnewTransition" within all  the python files in `C:\Users\codyj\Documents\KiCad\6.0\3rdparty\`  and its subfolders.

MitjaNemec, 2023-03-19:
So it look like this should be fixed in newer version of pcbnewTransition. You'll have to update pcbnew's python environment