MitjaNemec/PlaceFootprints #15: Fatal error when placing footprints

StayWavy1, 2023-04-21:
Getting this error when I run plugin with sheet nr.

I have a unit cell with some parts that I am trying to replicate in a 4x4 array/matrix.

Log file output:
04-20 21:45:13 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 405:Plugin executed on: 'win32'
04-20 21:45:13 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 406:Plugin executed with python version: '3.9.16 (main, Apr  9 2023, 05:22:13) [MSC v.1934 64 bit (AMD64)]'
04-20 21:45:13 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 407:KiCad build version: 7.0.2
04-20 21:45:13 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 408:Plugin version: 2.0.3
04-20 21:45:13 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 409:Frame repr: <wx._core.Frame object at 0x00000232A44D6940>
04-20 21:45:13 com_github_MitjaNemec_PlaceFootprints.place_footprints 121:getting a list of all footprints on board
04-20 21:45:13 com_github_MitjaNemec_PlaceFootprints.place_footprints 140:Footprint S1 does not have Sheetfile property, it will not be considered for placement. Most likely it is only in layout
04-20 21:45:13 com_github_MitjaNemec_PlaceFootprints.place_footprints 140:Footprint REF** does not have Sheetfile property, it will not be considered for placement. Most likely it is only in layout
04-20 21:45:20 com_github_MitjaNemec_PlaceFootprints.place_footprints 244:constructing a list of sheets suitable for replication on level:'VQ33 Unit Cell', file:'C:\\Users\\jdean\\Documents\\X\\Unit Cell.kicad_sch'
04-20 21:45:20 com_github_MitjaNemec_PlaceFootprints.place_footprints 279:suitable sheets are:[['HPOL', 'HPOL Quad 1', 'HQ11 Unit Cell'], ['HPOL', 'HPOL Quad 1', 'HQ12 Unit Cell'], ['HPOL', 'HPOL Quad 1', 'HQ13 Unit Cell'], ['HPOL', 'HPOL Quad 1', 'HQ14 Unit Cell'], ['HPOL', 'HPOL Quad 2', 'HQ21 Unit Cell'], ['HPOL', 'HPOL Quad 2', 'HQ22 Unit Cell'], ['HPOL', 'HPOL Quad 2', 'HQ23 Unit Cell'], ['HPOL', 'HPOL Quad 2', 'HQ24 Unit Cell'], ['HPOL', 'HPOL Quad 3', 'HQ31 Unit Cell'], ['HPOL', 'HPOL Quad 3', 'HQ32 Unit Cell'], ['HPOL', 'HPOL Quad 3', 'HQ33 Unit Cell'], ['HPOL', 'HPOL Quad 3', 'HQ34 Unit Cell'], ['HPOL', 'HPOL Quad 4', 'HQ41 Unit Cell'], ['HPOL', 'HPOL Quad 4', 'HQ42 Unit Cell'], ['HPOL', 'HPOL Quad 4', 'HQ43 Unit Cell'], ['HPOL', 'HPOL Quad 4', 'HQ44 Unit Cell'], ['VPOL', 'VPOL Quad 1', 'VQ11 Unit Cell'], ['VPOL', 'VPOL Quad 1', 'VQ12 Unit Cell'], ['VPOL', 'VPOL Quad 1', 'VQ13 Unit Cell'], ['VPOL', 'VPOL Quad 1', 'VQ14 Unit Cell'], ['VPOL', 'VPOL Quad 2', 'VQ21 Unit Cell'], ['VPOL', 'VPOL Quad 2', 'VQ22 Unit Cell'], ['VPOL', 'VPOL Quad 2', 'VQ23 Unit Cell'], ['VPOL', 'VPOL Quad 2', 'VQ24 Unit Cell'], ['VPOL', 'VPOL Quad 3', 'VQ31 Unit Cell'], ['VPOL', 'VPOL Quad 3', 'VQ32 Unit Cell'], ['VPOL', 'VPOL Quad 3', 'VQ34 Unit Cell'], ['VPOL', 'VPOL Quad 4', 'VQ41 Unit Cell'], ['VPOL', 'VPOL Quad 4', 'VQ42 Unit Cell'], ['VPOL', 'VPOL Quad 4', 'VQ43 Unit Cell'], ['VPOL', 'VPOL Quad 4', 'VQ44 Unit Cell']]
04-20 21:45:45 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 476:Sheets selected: [['VPOL', 'VPOL Quad 1', 'VQ11 Unit Cell'], ['VPOL', 'VPOL Quad 1', 'VQ12 Unit Cell'], ['VPOL', 'VPOL Quad 1', 'VQ13 Unit Cell'], ['VPOL', 'VPOL Quad 1', 'VQ14 Unit Cell'], ['VPOL', 'VPOL Quad 2', 'VQ21 Unit Cell'], ['VPOL', 'VPOL Quad 2', 'VQ22 Unit Cell'], ['VPOL', 'VPOL Quad 2', 'VQ23 Unit Cell'], ['VPOL', 'VPOL Quad 2', 'VQ24 Unit Cell'], ['VPOL', 'VPOL Quad 3', 'VQ31 Unit Cell'], ['VPOL', 'VPOL Quad 3', 'VQ32 Unit Cell'], ['VPOL', 'VPOL Quad 3', 'VQ34 Unit Cell'], ['VPOL', 'VPOL Quad 4', 'VQ41 Unit Cell'], ['VPOL', 'VPOL Quad 4', 'VQ42 Unit Cell'], ['VPOL', 'VPOL Quad 4', 'VQ43 Unit Cell'], ['VPOL', 'VPOL Quad 4', 'VQ44 Unit Cell']]
04-20 21:45:45 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 485:Footprints to place: ['C126', 'C33', 'C55', 'C77', 'C99', 'C66', 'C22', 'C44', 'C88', 'C142', 'C153', 'C115', 'C180', 'C191', 'C169', 'C207']
04-20 21:45:45 com_github_MitjaNemec_PlaceFootprints.place_footprints 398:Starting placing with matrix layout
04-20 21:45:45 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 561:Fatal error when executing place footprints
Traceback (most recent call last):
  File "C:\Users\jdean\Documents\KiCad\7.0\3rdparty\plugins\com_github_MitjaNemec_PlaceFootprints\action_place_footprints.py", line 557, in Run
    placer.place_matrix(sorted_footprints, ref_fp_ref, step_x, step_y, nr_columns, step, rotation, True)
  File "C:\Users\jdean\Documents\KiCad\7.0\3rdparty\plugins\com_github_MitjaNemec_PlaceFootprints\place_footprints.py", line 426, in place_matrix
    fp.fp.SetPosition(pcbnew.VECTOR2I(*new_position))
  File "C:\Program Files\KiCad\7.0\bin\Lib\site-packages\pcbnew.py", line 4826, in __init__
    _pcbnew.VECTOR2I_swiginit(self, _pcbnew.new_VECTOR2I(*args))
OverflowError: in method 'new_VECTOR2I', argument 1 of type 'int'

StayWavy1, 2023-04-23:
So I have been experimenting with this problem and found if I change the units to mm from mil (which is what I was working in), no error occurs. Guess with my board being pretty big and in mil it caused an overflow somehow?

StayWavy1, 2023-04-23:
Do you have any plans of making this feature like AutoCAD's array feature in the future? It would be nice if you could make a parent cell that controls all other (child) cells and locks & mimics the child cells down to reflect what is in the parent cell (even if parent is edited later). Could also have a feature to manually change one child cell if you needed something custom in one for whatever reason. 

I work a lot of phased arrays PCBs so this would be really useful for that application.

maciekish, 2024-05-21:
I have the same issue

```
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 647:Plugin executed on: 'darwin'
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 648:Plugin executed with python version: '3.9.13 (v3.9.13:6de2ca5339, May 17 2022, 11:37:23) \n[Clang 13.0.0 (clang-1300.0.29.30)]'
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 649:KiCad build version: 8.0.1
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 650:Plugin version: 3.0.0
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.action_place_footprints 651:Frame repr: <wx._core.Frame object at 0x16dd8faf0>
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 121:getting a list of all footprints on board
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint H4 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint H2 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint J1 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint H1 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C20 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint L2 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C7 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint R10 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint SW4 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C21 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C28 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint R3 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C6 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C10 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint H3 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint Y1 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint R4 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint AE1 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint WARNING on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C23 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint R6 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint L3 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C9 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint SW2 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint Q2 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C27 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint MD1 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C19 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint R5 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint UART1 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint DRX2 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint D2 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint R9 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C2 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C8 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint DPWR2 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint DRX1 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint D3 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint R2 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint SPI1 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C14 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C12 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C11 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C15 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint L1 on root level
05-22 00:48:50 com_github_MitjaNemec_PlaceFootprints.place_footprints 156:Footprint C25 on root level
[... cut ...]
    placer.place_matrix(sorted_footprints, ref_fp_ref, step_x, step_y, nr_columns, step, rotation, True)
  File "/Users/maciekish/Documents/KiCad/8.0/3rdparty/plugins/com_github_MitjaNemec_PlaceFootprints/place_footprints.py", line 453, in place_matrix
    fp.fp.SetPosition(pcbnew.VECTOR2I(*new_position))
  File "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pcbnew.py", line 5001, in __init__
    _pcbnew.VECTOR2I_swiginit(self, _pcbnew.new_VECTOR2I(*args))
OverflowError: in method 'new_VECTOR2I', argument 1 of type 'int'
```

MitjaNemec, 2024-05-22:
Thanks to both for reporting this. I am a bit short on time lately, so please have patience.

@maciekish do you work in mils(inches) or in millimeters? If you work in mils, does it help if you switch to millimeters?

@maciekish, @StayWavy1 can anybody share a project or prepare a test project so that I can recreate the issue locally?

maciekish, 2024-05-22:
Seems like the issue is mils indeed. If i switch to mm it works.