MitjaNemec/PlaceFootprints #18: Crashes when no value entered

tomek-szczesny, 2023-07-22:
Hi again,

The script crashes when the "step y" field is empty (the only thing I tested).
It crashes somewhere where commas are replaced by dots I think.
Looks like a matter of sanity check, if the field is empty then it should be filled with "0".

By the way all fields could be empty by default, so it's easier to do the same thing multiple times by pasting values on each run.

MitjaNemec, 2023-07-31:
Thanks for reporting.

I'll look into if I can save previously used values within a particular project

tomek-szczesny, 2023-07-31:
That would be awesome, for now I modified the default values in the script when I had 40 components to place. :)
Perhaps those values per project could be stored in a text box on some User Drawing layer? 
Then the user could delete it if they are bothered with it, or copy it to another PCB if they want to.

MitjaNemec, 2023-08-02:
Well I'll either store the settings as a dedicated file in the project folder which would make the editing simple. But to be honest I'd rather store them in OS temporary folder, as some users with extreme OCD like to keep the project folder tidy. And they will likely complain if I start messing up the project folder or godforbid even the project itself (User Drawing layer)

MitjaNemec, 2023-08-06:
So, I've added:
- the feature which saves/restores used configuration
- prevents running the plugin on missing entries, and highlights missing entries.

You can check the 2.1.1 release by manually installing from file, as it will be a while before it is updated in PCM

tomek-szczesny, 2023-08-09:
Okay I got an extra assignment so I updated the plugin to 2.1.1.

Seems like there might be a regression unfortunately. Now when I want to place footprints by sheet number, the "sheets to replicate" box is empty, and plugin does nothing. 

The step value is not saved either, I cannot see an extra file being created anywhere.

Same applies to version 2.1.0.

Back to version 2.0.3 and everything works fine again.

Please let me know if I can help debug this. I work on Manjaro Linux if that matters.

MitjaNemec, 2023-08-11:
Thanks for reporting this. It should be fixed in 2.1.2

tomek-szczesny, 2023-08-17:
Hi,
I'm testing 2.1.2 right now.
The regression is gone, the tool is working again. For any reason I have to use negative X coordinate to place footprints on the right, but that might be an issue with sheet enumeration in this project.

It still doesn't remember settings, and I can't see any file created in project directory, though.

MitjaNemec, 2023-10-13:
Repoening, so that I don't forget that:
>It still doesn't remember settings, and I can't see any file created in project directory, though.
in the per sheet mode

MitjaNemec, 2023-10-13:
@tomek-szczesny, I've tested the plugin throughly and I did not have any issues with saving the config either in "per reference" or in "per sheet" mode.

Can you update to [2.1.3](https://github.com/MitjaNemec/PlaceFootprints/releases/tag/2.1.3) and retest the plugin. If you notice any issues please remember to store the .log file and attach it here. Thanks