MitjaNemec/PlaceFootprints #19: Does not work on nightly 7.99

zhangyh89, 2023-09-16:
09-16 15:47:05 ReplicateLayout.action_replicate_layout 359:Plugin executed on: 'win32'
09-16 15:47:05 ReplicateLayout.action_replicate_layout 360:Plugin executed with python version: '3.11.2 (main, Aug  9 2023, 09:40:36) [MSC v.1936 64 bit (AMD64)]'
09-16 15:47:05 ReplicateLayout.action_replicate_layout 361:KiCad build version: 7.99.0-2768-g0df78f2f68
09-16 15:47:05 ReplicateLayout.action_replicate_layout 362:Plugin version: 2.0.7
09-16 15:47:05 ReplicateLayout.action_replicate_layout 363:Frame repr: <wx._core.Frame object at 0x000002961502A560>
09-16 15:47:05 ReplicateLayout.action_replicate_layout 387:Preparing replicator with D1 as a reference
09-16 15:47:05 ReplicateLayout.replicate_layout 113:getting a list of all footprints on board
09-16 15:47:05 ReplicateLayout.action_replicate_layout 402:Fatal error when making an instance of replicator
Traceback (most recent call last):
  File "C:\Users/xxx/Documents/KiCad/7.99/scripting\plugins\ReplicateLayout\action_replicate_layout.py", line 391, in Run
    replicator = Replicator(board, src_anchor_fp_reference)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users/xxx/Documents/KiCad/7.99/scripting\plugins\ReplicateLayout\replicate_layout.py", line 129, in __init__
    sheet_file = fp.GetProperty('Sheetfile')
                 ^^^^^^^^^^^^^^
AttributeError: 'FOOTPRINT' object has no attribute 'GetProperty'

MitjaNemec, 2023-09-25:
The plugin only works on KiCad version 7.0.x. Version 7.99 is currently not supported.

MitjaNemec, 2024-01-26:
The V8 compatible plugin is currently available at: https://github.com/MitjaNemec/PlaceFootprints/releases/tag/3.0.0 .

closed 2024-01-26 by MitjaNemec