MitjaNemec/PlaceFootprints #20: Crashes in Kicad-7.0.9

stormyweathers, 2023-12-26:
I'm running kicad-7.0.9 on ubuntu-20.0.4 and I can't get this plugin to work.

[image]

These are the contents of `place_footprints.log`
```
 12-26 15:14:38 place_footprints.action_place_footprints 414:Plugin executed on: 'linux'
 12-26 15:14:38 place_footprints.action_place_footprints 415:Plugin executed with python version: '3.8.10 (default, Nov 22 2023, 10:22:35) \n[GCC 9.4.0]'
 12-26 15:14:38 place_footprints.action_place_footprints 416:KiCad build version: 7.0.9-7.0.9~ubuntu20.04.1
 12-26 15:14:38 place_footprints.action_place_footprints 417:Place footprints plugin version: 54 started
```

MitjaNemec, 2023-12-27:
Hi, thanks for reporting this. 

It seem that you are trying to use really old version of the plugin.
So I think your issues would be resolved if you used latest version of the plugin compatible with KiCad 7.0.x. You can get it through KiCad's PCM (plugin and content manager)

BTW, where did you get the plugin? How did you install the plugin?

stormyweathers, 2023-12-28:
I installed the plugin through the kicad plugin manager, using the official kicad repository.
I have version 2.1.3 of your plugin.

MitjaNemec, 2023-12-28:
something does not make sense.

In 2.1.3 version of the plugin, the 427 line of the [action_place_footprints.py](https://github.com/MitjaNemec/PlaceFootprints/blob/d15ebca690fa37b761ae606379c195fb478cc0e8/action_place_footprints.py#L427) does not match your screenshot. Also The .log file contends does not agree with what version 2.1.3 should produce. I am attaching an example for 2.1.2, but 2.1.3 should generate something similar. Please note the line numbers
```
10-13 11:05:35 PlaceFootprints.action_place_footprints 644:Plugin executed on: 'win32'
10-13 11:05:35 PlaceFootprints.action_place_footprints 645:Plugin executed with python version: '3.9.16 (main, Aug  8 2023, 05:38:28) [MSC v.1936 64 bit (AMD64)]'
10-13 11:05:35 PlaceFootprints.action_place_footprints 646:KiCad build version: 7.0.8
10-13 11:05:35 PlaceFootprints.action_place_footprints 647:Plugin version: 2.1.2
```

Did you by any chance manually install the plugin some time in the past, before the PCM. You might have placed the plugin in the folder that would not be overwritten by new installation by the PCM. Thus you might actually have two installed versions of the plugin, and KiCad loads the old one.

stormyweathers, 2023-12-28:
You are correct.  I must have manually installed long ago and forgotten about it.  After manually purging and reinstalling through the PCM, it works!

Thanks!

closed 2023-12-28 by stormyweathers