MitjaNemec/PlaceFootprints #22: Running Plugin raises AttributeError: 'float' object has no attribute 'AsTenthsOfADegree'

dm2lct, 2024-03-27:
Under Kicad 6.0.11 under Debian 12 the plugin seems not to work.

It raises a fatal error when trying to run it:

  * I selected a Footprint from a hierachical schematic
  * opened the plugin config and set it to matrix and 6 columns
  * left everything else untouched
  * ran it

Then the attached error message came. 

In the log there is a python error raised.

The same issue occoured when trying to tun the replicate layout plugin.

I hope this is enough for describing the error. 

Thanks a lot for your work!

[image]

[image]

[place_footprints.log](https://github.com/MitjaNemec/PlaceFootprints/files/14773547/place_footprints.log)

MitjaNemec, 2024-03-27:
Yeah, that is the same as issue #14 and [#47](https://github.com/MitjaNemec/ReplicateLayout/issues/47) and [#56](https://github.com/MitjaNemec/ReplicateLayout/issues/56) in the ReplicateLayout plugin.

The bug is within "PcbnewTransition", which was installed as a dependency of some other plugin. Either update "PcbnewTransition" or remove it.

closed 2024-03-27 by MitjaNemec

dm2lct, 2024-03-27:
Thanks a lot! 
After I deinstalled KiKit and deleted ./.local/lib/python3.11/site-packages/pcbnewTransition
everything works perfectly.