MitjaNemec/PlaceFootprints #23: Issue Placing Footprints

HayHay116, 2024-04-30:
[image]

Running KiCad 7.0.10 on Mac Sonoma 14.0

The issue arises after selecting a footprint and then running the plugin. Attached is a picture of the prompt and the log.
[place_footprints.log](https://github.com/MitjaNemec/PlaceFootprints/files/15169642/place_footprints.log)

MitjaNemec, 2024-05-06:
Yeah, you are running 3.x version of the plugin on a KiCad 7.x. 

You should use plugin version 2.x, which are compatible with KiCad 7.x
Plugin version 3.x are compatible with KiCad 8.x

How did you install the plugin. It should not be possible to install 3.x plugins in KiCad 7.x, by default. Only if you tick the "Show all versions" in PCM you can select and install non compatible version

[image]

MitjaNemec, 2024-05-06:
BTW the above screenshot was made in KiCad 8.0.2

insurgus, 2024-10-05:
Is it possible you set "Rorate every n-th footprint" to a value of 0? There's a bug where this causes a fatal crash.