MitjaNemec/PlaceFootprints #24: Issue with placing footprints in a circle

marc-cr1810, 2024-07-17:
No matter what the radius is set to, the plugin will ignore it and just use 0.

I am using KiCad v8.0

MitjaNemec, 2024-07-29:
Hmmm, on what OS are you? What is your locale environments (OS language settings, KiCad language settings, units used)?

huynguyen1989, 2024-08-31:
I think he mentioned about the parameter 
`Rotate every n-th footprint = 0` Will cause the error

[image]

marc-cr1810, 2024-08-31:
> Hmmm, on what OS are you? What is your locale environments (OS language settings, KiCad language settings, units used)?

Just Windows 11, English for both and metric mm/cm

marc-cr1810, 2024-08-31:
> I think he mentioned about the parameter 
> `Rotate every n-th footprint = 0` Will cause the error
> 
> [image]
> 

No, I did rotate every 15, and it did rotate them, but the positions were all dead centre and not placed at a distance from the centre

ciriousjoker, 2025-04-02:
Fails on mac os for me:

[place_footprints.log](https://github.com/user-attachments/files/19575978/place_footprints.log)

Trying to lay out 28 leds in a circle