MitjaNemec/PlaceFootprints #27: Support Kicad 9 ?

Witawat, 2025-02-21:
How to use on Kicad 9 ?

UrbanCircles, 2025-03-25:
would be great to have this on kicad 9!!

3lefeint, 2025-03-25:
i found a way how you can use the plugin with kicad 9.0. But i did not test it thoroughly so i cant say if there are new issues.

but open the plugin manager, select Place Footprints. Then tick Show all versions - not only "compatible" - and then select the latest and install it. it works for me

[image]

MitjaNemec, 2025-04-02:
Yeah, it tuns out that the plugin is compatible with V9 as is, but I way cautious and did not enable the support by default.

I've released V9 compatible version and it should be available in PCM in a day or so.

@Witawat , @3lefeint , @UrbanCircles can I ask you what does this plugin give you compared to footpirnt placement options native to KiCad?

closed 2025-04-02 by MitjaNemec