MitjaNemec/PlaceFootprints #28: Fatal error when placing footprints in a circle

ciriousjoker, 2025-04-02:
Here's the log, trying to place a bunch of leds in a circle.

[place_footprints.log](https://github.com/user-attachments/files/19575992/place_footprints.log)

ciriousjoker, 2025-04-02:
Seems like it was caused because I had a random other status led. No idea why that caused an issue, but with the correct LED selection everything works finde.

MitjaNemec, 2025-04-04:
Yeah, the plugin also places the footpirnt text properly and before doing this it checks that footprint has the same number of text items as selected footprint. It seems that one of the footprints was different in this regard. The plugin should emit user friendly message but due to the bug it crashed.

thanks for reporting this, I'll fix the bug

commit 731e3fed (2025-04-04): Fix a bug in error message

Fixes #28
closed 2025-04-04 by MitjaNemec